dmadison/ArduinoXInput_Teensy #19: Teensy 4.1 Support

jrsmile, 2020-06-26:
With the release of the Teensy 4.1 it is easy to hook up a ps4 controller to it by just adding a usb-A socket to the provided pins.
would you be so kind and add teensy 4.1 support for your modification?

gdsports, 2020-08-20:
Teensy 3.6 has similar USB host support so it could be used instead. But it would be nice to have a 600 MHz gamepad! Voice control via TensorFlow Lite?

shonenrek, 2021-05-08:
I would also like to see this, there are many possibilities but even having a controller that's able to work plug-and-play on the 4.0 or 4.1 would be great.

SantiagoSaldana, 2021-11-07:
Just wanted to bump and see if any progress has been made on this, I've spent the last few days trying to port it over from teensy 3.6 -> teensy 4.1 and there is quite a jump in complexity.  Using the nsGadget_Teensy library for reference as they have there's working on Teensy 3.x/4.x

mdluffy1234, 2022-06-14:
4.1 support needs to happen ASAP cuz due to the chip shortage, there won't be any more 3.6s until June 2023, and the 4.1 just so happens to not be affected by that shortage.

kje123, 2022-07-03:
Sorry to bump but have you made any progress since your last post? @SantiagoSaldana

SantiagoSaldana, 2022-07-03:
> 

Unfortunately no, I finished my project by buying a 3.6.  I ran into too many issues trying to port and had to give up on it.

wheybags, 2022-08-07:
I had a go at this, it's working well for me: https://github.com/dmadison/ArduinoXInput_Teensy/pull/26

mdluffy1234, 2022-08-30:
> I had a go at this, it's working well for me: #26 on a 4.1.
> 
> If anyone here could test it on a 4.0 that would be cool!

What version did you use?

wheybags, 2022-08-30:
I used a teensy 4.1.

mdluffy1234, 2022-09-02:
> I used a teensy 4.1.

Sorry, I meant the software.

wheybags, 2022-09-02:
This repo readme says it's a patch on teensyduino v1.56, so I made my patches on top of that version too.

dmadison, 2022-10-30:
Thanks to @wheybags, support has been added for all Teensy 4 boards including the Teensy 4.0, Teensy 4.1, and Teensy MicroMod :grin:

closed 2022-10-30 by dmadison

Shreyash170304, 2024-08-01:
> With the release of the Teensy 4.1 it is easy to hook up a ps4 controller to it by just adding a usb-A socket to the provided pins. would you be so kind and add teensy 4.1 support for your modification?

hey, will you plz send me the resources for connecting PS4 with teensy 4.1. I had tried my best but i got failed, it will be very helpful if you do so if you have some resources.