dmadison/ArduinoXInput_Teensy #30: No longer compiles in PlatformIO for teensy 3.2

RootUser3885, 2023-02-13:
I ran into an issue where the XInput library just stopped compiling.
I had compiled it before with no issue.
The project compiles just fine if using USB_SERIAL mode, but wont as soon as I set it to USB_XINPUT.

I re-opened a test project I used to test the library that I know for a fact compiled previously.

I don't know what changed. I even used another computer that I had previously compiled this on.

The terminal shows this:
P:\Source Code\PlatformIO\Active\XInput Basic Test\.pio\libdeps\teensy31\XInput\src\XInput.cpp: In constructor 'XInputController::XInputController()':
P:\Source Code\PlatformIO\Active\XInput Basic Test\.pio\libdeps\teensy31\XInput\src\XInput.cpp:204:2: error: 'XInputUSB' has not been declared
  XInputUSB::setRecvCallback(XInputLib_Receive_Callback);
  ^
P:\Source Code\PlatformIO\Active\XInput Basic Test\.pio\libdeps\teensy31\XInput\src\XInput.cpp: In member function 'boolean XInputController::connected()':
P:\Source Code\PlatformIO\Active\XInput Basic Test\.pio\libdeps\teensy31\XInput\src\XInput.cpp:435:9: error: 'XInputUSB' has not been declared
  return XInputUSB::connected();
Compiling P:\Source Code\PlatformIO\Active\XInput Basic Test\.pio\build\teensy31\FrameworkArduino\HardwareSerial2.cpp.o
         ^
P:\Source Code\PlatformIO\Active\XInput Basic Test\.pio\libdeps\teensy31\XInput\src\XInput.cpp: In member function 'int XInputController::send()':
P:\Source Code\PlatformIO\Active\XInput Basic Test\.pio\libdeps\teensy31\XInput\src\XInput.cpp:446:9: error: 'XInputUSB' has not been declared
  return XInputUSB::send(tx, sizeof(tx));
         ^
P:\Source Code\PlatformIO\Active\XInput Basic Test\.pio\libdeps\teensy31\XInput\src\XInput.cpp: In member function 'int XInputController::receive()':
P:\Source Code\PlatformIO\Active\XInput Basic Test\.pio\libdeps\teensy31\XInput\src\XInput.cpp:455:6: error: 'XInputUSB' has not been declared
  if (XInputUSB::available() == 0) {
      ^
P:\Source Code\PlatformIO\Active\XInput Basic Test\.pio\libdeps\teensy31\XInput\src\XInput.cpp:461:24: error: 'XInputUSB' has not been declared
  const int bytesRecv = XInputUSB::recv(rx, sizeof(rx));
                        ^

The project is just a simple single button thats always false.

[platformio.ini.txt](https://github.com/dmadison/ArduinoXInput_Teensy/files/10722975/platformio.ini.txt)
[main.cpp.txt](https://github.com/dmadison/ArduinoXInput_Teensy/files/10722983/main.cpp.txt)

dmadison, 2023-02-13:
It sounds like you have updated the Teensy core and forgot to add the new XInput modifications. The errors are because it cannot find the `XInputUSB` name, which would make sense if that file doesn't exist.

Have you tried compiling with the Arduino IDE? Is the 'XInput' USB option there?

RootUser3885, 2023-02-14:
Yes, it compiles in the Arduino IDE with the XInput option selected. I had suspected that so I checked in the .platformio/packages/framework-arduinoteensy/cores/teensy3/ folder. usb_xinput.cpp and .h are there. I compared usb-desc and the XInput modifications are there. Same for teensy4. Is there a path I am missing?

[image]

[teensy3.zip](https://github.com/dmadison/ArduinoXInput_Teensy/files/10727555/teensy3.zip)

dmadison, 2023-02-14:
Hmm.

The next step is probably to enable the full compilation output and then look through the results. It's possible that the Teensy XInput files are missing from wherever it's compiling from (maybe a cache?), or that because they're extra they're not included in some prefab'd Makefile. I'm not sure.

I don't use PlatformIO myself, and the project does not officially support PlatformIO due to the additional changes required for the meta build files. If it compiles in the Arduino toolchain then I'd suggest using that (IDE or CLI).

RootUser3885, 2023-02-14:
Oh, Sorry. I didn't realize the project didn't support PlatformIO. I found it within platformIO itself. https://registry.platformio.org/libraries/dmadison/XInput/installation

Ok, since the library doesn't officially support platformio then I guess this issue can be closed.

In the meantime I will mess with it a bit longer before switching to the Arduino IDE. (I will miss intellisense).

If I can figure something out I will comment on this for future reference and others.

Thank you