dmadison/ArduinoXInput_Teensy #33: Cryptic Error when trying to use Xinput Library with Teensy 3.6

DNCrosby, 2023-04-26:
I get the following when I try and setup the Teensy 3.6, I have a 4.1 board here but I read that needs a different library or patch, so I'm hesitant to try it, if I have problems at this stage.

Error resolving FQBN: getting

Error compiling for board Teensy 3.6.

I'm using Arduino 1.8.19 and Teensyduino 1.56 (With the driver from https://github.com/dmadison/ArduinoXInput_Teensy/releases/tag/v1.0.12 ) Which reads as compatible...

My Goal is to use a home-made joystick with a MiiSter board, and I tried HID and it didn't recognize it as a Joystick, once it got into the game. While the Xbox controller I attached worked fine. So this is my alternative to try and see if I can get a driving game with analog inputs to work on things like Outrun and Spy Hunter.

dmadison, 2023-04-26:
Hello, sorry to hear that you're having problems. The error you're receiving isn't clear but it's likely something went wrong during installation. Please enable verbose compilation in the IDE (`Preferences -> Show verbose output during: Compilation`) and paste the full console output here.

Please also provide some more information:
* What operating system are you using?
* What are all of board options are selected in the IDE? (CPU speed, optimize, etc.)
* Are you compiling directly from the IDE interface (GUI) or from a script?

If you are trying to get your 4.1 board working I would recommend using the latest version of the repo instead.

DNCrosby, 2023-04-27:
So I found it the verbose output helped, the Arduino IDE is Arduino 1.x yet it was looking in C:\Users\[USERNAME]\AppData\Local\Arduino15\packages\teensy\hardware\ instead of the C:\Program Files (x86)\Arduino\hardware\teensy\avr, unless I read that wrong on your install instructions. But during the verbose part I could see what folder it was looking in. 
I do also have a version of 2x installed... so maybe that's why...
Thanks for the help! :D

dmadison, 2023-04-27:
If you have both versions of the IDE installed and two Teensy installations that will cause issues.

I would recommend staying away from the 2.x IDE for the time being as it's quite buggy, especially if you're making modifications to the boards files.

dmadison, 2023-05-06:
I'm assuming this is resolved. Feel free to reopen if you still need more help.

closed 2023-05-06 by dmadison