dmadison/ArduinoXInput_Teensy #34: "Xinput" option does not appear in USB types for Teensy 4.1

nupanick, 2023-05-10:
"Xinput" option does not appear in USB types for Teensy 4.1

It does appear for Teensy 4.0, though.

Tested with Arduino 1.8.19 and 2.1.0, same results on both. 

Teensyduino is version 1.58.1.

dmadison, 2023-05-10:
Hello. Sorry to hear you're having trouble. I can't replicate your problem.

I recommend users stay away from the 2.x IDE for the time being as it has a bug which prevents modifying boards files.

Please try the following:
* Uninstall the 2.1.0 IDE.
* Navigate to `%AppData%\..\Local\Arduino15\packages\` and delete the `teensy` folder if there is one.
* Delete your `teensy` hardware folder in your 1.x IDE installation
* Reinstall Teensyduino for the 1.x IDE
* Follow the XInput installation instructions for the 1.x IDE

nupanick, 2023-05-10:
Thank you! We had to completely delete and reinstall the files but now it works.

closed 2023-05-10 by nupanick